TheThingsNetwork/docs #24: Automate pulling existing content from repos

FokkeZB, 2016-09-19:
1. Generate, e.g. for TTNCTL with https://github.com/spf13/cobra/blob/master/doc/man_docs.md
2. Pull, e.g. https://github.com/TheThingsNetwork/ttn/blob/refactor/mqtt/README.md for MQTT

This should help us in particular with API reference to keep the API reference with the source but still have it available through docs as well.

It would require:

- The MD pulled in to be just that; API reference
- A script and config file with files to pull and where to put
- Run the script on every docs publish to update

htdvisser, 2016-09-19:
1: we used to do this for the wiki too, see for example https://www.thethingsnetwork.org/wiki/Backend/ttnctl/ttnctl

FokkeZB, 2016-09-19:
@htdvisser I know, and I plan on moving that into the official docs.

FokkeZB, 2016-09-19:
2) is done by https://github.com/TheThingsNetwork/docs/commit/fc506be16d709f142f1123bb1af142ee65d3833c:  https://github.com/TheThingsNetwork/docs#automatic-updates

closed 2016-09-19 by FokkeZB