TheThingsNetwork/docs #32: Guide: Java Library

FokkeZB, 2016-09-19:


cambierr, 2016-09-19:
I started writing it but troubles with my local setup (style not loaded so it's hard to get a real preview) made me not able to commit/push anything clean. I'll complete it asap

FokkeZB, 2016-09-19:
Mmm, interesting. It should work if you follow the steps from the readme. What errors did you run in to? Could you create an issue at https://github.com/thethingsNetwork/docs/issues ?

cambierr, 2016-09-19:
From what I see, there is no error, just a missing main.css file.
I'm running it inside a docker container to avoid installing tons of package on my machine. I'll investigate that later this week and create an issue in case I can't manage to have it working

FokkeZB, 2016-09-20:
Jekyll should generate that file at `_site/css/main.css` when it generates the site. Let me know.

closed 2016-11-07 by FokkeZB