TheThingsNetwork/docs #54: Rename "refactor" to "v2-preview"

htdvisser, 2016-10-06:


FokkeZB, 2016-10-06:
@htdvisser I thought about `preview` to be better than `refactor`, but does this mean we will rename `staging.thethingsnetwork.org` to `v1-preview.thethingsnetwork.org`? That would make it more logical.

htdvisser, 2016-10-06:
I'm afraid it's a little too late for that

FokkeZB, 2016-10-07:
Cant we just redirect staging.ttn to preview-v1.console.ttn and preview.console to preview-v2.console.ttn? It's hard to tell what comes first, staging or preview

htdvisser, 2016-10-07:
I don't know how much work that will be. @johanstokking, @romeovs and me will discuss the (technical) migration process from staging to the new backend today. I think that spending a lot of time on renaming staging to preview-v1 is not reasonable, but when it can be done quickly it might be worth it.

johanstokking, 2016-10-07:
Let's keep staging as-is, remove it soon from docs, let it die silently and take it offline in 2017Q1.

commit 3a01f415 (2016-10-07): Closes #54, Closes #53, Fixes #35, Adds back to top link

Rename refactor to v2-preview, redirect from refactor to v2-preview, add support for guide aliases to link dashboard to console, rename dashboard to console, make sidebar scrollable, add back to top link
closed 2016-10-07 by FokkeZB