TheThingsNetwork/docs #62: Improvement: Use byte pointers instead of bit shifting?

FokkeZB, 2016-10-13:
We can get rid of the bit shifting on Arduino by using byte pointers instead. This might be easier for users to understand:

Before:

```c
byte payload[4];

long lng = 200000L;
bytes[0] = (int) ((lng & 0xFF000000) >> 24 );
bytes[1] = (int) ((lng & 0x00FF0000) >> 16 );
bytes[2] = (int) ((lng & 0x0000FF00) >> 8  );
bytes[3] = (int) ((lng & 0X000000FF)       );
```

After:

```c
byte payload[4];

long lng = 200000L;
byte* bytes = (byte*) &lng;
bytes[0] = bytes[3];
bytes[1] = bytes[2];
bytes[2] = bytes[1];
bytes[3] = bytes[0];
```

What do you think @johanstokking?

johanstokking, 2016-10-13:
The last example is not going to work, I suppose you mean `payload[x]` on the left side.

But yes, we can do this. 

As discussed, we need to explicitly state the length of the type. Instead of `long`, we need `uint32_t` or `int32_t`. Same for `int` and `byte`.

FokkeZB, 2016-10-13:
Fixed the example.

I'll document that users should use explicit length and I have created https://github.com/TheThingsNetwork/arduino-device-lib/issues/103 to do so in the examples.

closed 2018-01-09 by johanstokking