TheThingsNetwork/docs #112: Flowchart: Downlink windows

FokkeZB, 2016-12-16:
Create a graphic to show that:

- Only last scheduled downlink is preserved
- You have 2 windows of opportunity to respond to deliver a downlink in response to a message you process.

/cc @wesselversluis @laurensslats

FokkeZB, 2016-12-16:
I think @johanstokking has one pretty good one in his slide deck?

johanstokking, 2016-12-20:
I don't have any unfortunately. But indeed, this is a handy graph

htdvisser, 2016-12-20:
[image]

htdvisser, 2016-12-20:
> @FokkeZB: Only last scheduled downlink is preserved

This will change in the future (https://github.com/TheThingsNetwork/ttn/issues/357)

> @FokkeZB: You have 2 windows of opportunity to respond to deliver a downlink in response to a message you process.

Not true. Only one.

FokkeZB, 2016-12-21:
Or this one we used in Ulm?

[image]

FokkeZB, 2016-12-21:
@htdvisser why only 1 opportunity? If you don't make the first window you get another chance the second one right?

htdvisser, 2016-12-21:
Because the Broker selects a downlink option before it forwards the message to the Handler. Therefore the downlink window has already been decided before a message is sent to the Application. Our Handler currently waits for a response from the application for max. 100ms, regardless of the downlink window.

closed 2020-05-12 by laurensslats