TheThingsNetwork/docs #146: Fix edit button

htdvisser, 2017-05-02:
On the website the edit button sometimes doesn't send the user to the correct page

johanstokking, 2018-02-15:
We should get rid of the edit button

htdvisser, 2018-02-15:
I disagree. Especially when we move the Wiki contents to the Docs, it shouldn't be too complicated to edit the docs. The Edit button (usually) sends you to the page on Github where you can edit and submit a pull request, which is great. Let's not remove that

johanstokking, 2018-02-15:
OK, I was thinking of inline editing, but it should be GitHub, so if "edit" is essentially a link then that's perfect.

What's wrong about the link?

htdvisser, 2018-02-19:
It sometimes didn't send you to the right page. Can't reproduce it right now. Will reopen when I find an example.

closed 2018-02-19 by htdvisser