TheThingsNetwork/docs #154: Unbrick Node docs

laurensslats, 2017-05-18:


jpmeijers, 2017-05-18:
Open up node, press reset button twice to get into bootloader mode. The LED should fade on and off. A USB TTY device should be seen on your computer now. To erase the user code from the Node, keeping only the bootloader in place, run:
`avrdude -patmega32u4 -cavr109 -P/dev/ttyACM0 -b57600 -e`

Or while the LED is fading on and off, program it with new code via the Arduino IDE. Sometimes the IDE is too slow and the Node exits bootloader mode and go back into the "bricked" state. The erase command is therefore advised.

johanstokking, 2017-05-18:
@jpmeijers please copy/paste this to the docs

jpmeijers, 2017-05-22:
https://www.thethingsnetwork.org/docs/devices/node/troubleshooting.html

closed 2017-05-22 by jpmeijers