TheThingsNetwork/docs #157: Error when using go-app-sdk

jpmeijers, 2017-06-23:
A couple of people reported getting:
```
go build main.go 
# github.com/TheThingsNetwork/ttn/api/protocol/lorawan
../../TheThingsNetwork/ttn/api/protocol/lorawan/message_conversion.go:101: cannot use m.DevNonce (type types.DevNonce) as type lorawan.DevNonce in assignment
../../TheThingsNetwork/ttn/api/protocol/lorawan/message_conversion.go:122: cannot use m.AppNonce (type types.AppNonce) as type lorawan.AppNonce in assignment
../../TheThingsNetwork/ttn/api/protocol/lorawan/message_conversion.go:123: cannot use m.NetId (type types.NetID) as type lorawan.NetID in assignment
```
When trying to use the go-app-sdk.

I think the sollution is running `govender sync`, but for a newbie to golang, this is not clear. Perhaps we can update the [Quick Start Guide](https://www.thethingsnetwork.org/docs/applications/golang/quick-start.html) to include something about `govender` under the setup section.

gomezjdaniel, 2017-06-23:
I got this error also, I fixed it by lowering versions of `github.com/brocaar/lorawan` and `github.com/brocaar/lorawan/band` packages. Remove them from your vendor file and add

```
		{
			"checksumSHA1": "ZHpBCsUv5lUxdt1gF9HWRZ4IffI=",
			"path": "github.com/brocaar/lorawan",
			"revision": "c61721fa96c85c25ea7ba635fc477224344ddbe3",
			"revisionTime": "2017-03-20T08:06:39Z"
		},
		{
			"checksumSHA1": "SlDtchaAX7SV5nD/BFP4x6fnmVk=",
			"path": "github.com/brocaar/lorawan/band",
			"revision": "c61721fa96c85c25ea7ba635fc477224344ddbe3",
			"revisionTime": "2017-03-20T08:06:39Z"
		},
```

jpmeijers, 2017-06-23:
What and where is "your vendor file"?

johanstokking, 2017-06-23:
I cannot reproduce. With `govendor sync` I get the vendored libs and tests work fine.

johanstokking, 2017-06-23:
Also report in the concerning repository if this is really an issue.

PR #159 (2017-06-23): Go minimum version. Govendor.
Fixes #158 and fixes #157
closed 2017-06-23 by johanstokking