TheThingsNetwork/docs #161: Gateway documentation overhaul

dotpy3, 2017-07-05:
The gateway owner experience is very poor at the moment. Configuring and maintaining a gateway is still painful, and there are ways in which the gateways documentation could be improved:

+ A leaner experience in the docs in general:
  + Introductions on gateways, how they work
  + A general glossary on common terms or concepts
+ How to buy gateways, or how to [build them](https://github.com/ttn-zh/ic880a-gateway) for enthusiasts. The current documentation only includes documentation for IoT Station, Conduit, Lorrier - but there are now many more LoRa gateways available. What models are currently available? What kind of usage are they destined for?
+ More details on packet forwarders - for every packet forwarder:
  + General information on the packet forwarder
  + Basic installation (gateway-agnostic)
  + How to configure it
+ More details on how to configure gateways, especially in regards to private networks.
I think the documentation *per gateway* should separated in these parts:
  + General hardware information: specs, how to start it, what are the important parts (antenna, buttons, information LEDs)
  + General software information: what's the OS powering the gateway? What's possible/not possible on them? Information on connecting to the gateway to manage it (SSH connection, serial port).
  + How to install the packet forwarder on a gateway
  + Advanced configuration options for a gateway

cc @ajovertoom

nullnvoids, 2017-07-05:
Exactly the information I would expect to be already present. The current work around is to run dedicated technical workshops breaking down each of the components. I would add to the above list  - backhaul options,  survey/scanning options, security hardening of both the Gateways and endpoint devices that are to be registered.

dotpy3, 2017-07-18:
I've started some work in that direction in the [feature/gateways-overhaul branch](https://github.com/TheThingsNetwork/docs/tree/feature/gateways-overhaul).
For the moment, I've added more information in the gateways introduction, information on what commercial gateways are available, where to find guides to build one, and an explanation on how they connect to the back-end.

Next steps before it can be considered done:
+ [ ] More details on the different packet forwarders
+ [ ] Overhaul of gateway configuration
+ [ ] Gateway debugging basics

johanstokking, 2017-07-24:
Is this closed since #163 ?

dotpy3, 2017-07-24:
Before closing this, I'd like to add more documentation *inside* individual gateways' documentation regarding the different between the packet forwarder.

closed 2018-01-09 by johanstokking