TheThingsNetwork/docs #176: Device initialisation

mazareth, 2017-11-15:
Hi,
I've a problem to add a device in my the things network account.
When a do that:
[image]

I've this error:
[image]

johanstokking, 2017-11-15:
Hi @mazareth, this means that the AppEUI and DevEUI combination already exists.

We clarified the error message in a PR that is open / not yet deployed.

closed 2017-11-15 by johanstokking

mazareth, 2017-11-15:
but that's because my device is a finished product
I can't change these parameters
How can i add my smilio in "the thing network"?

johanstokking, 2017-11-15:
Please contact the manufacturer in that case.

The problem is that it seems that application `demo-smiley` already exists with this device registered in it. You cannot take ownership of a AppEUI/DevEUI combination if it is already registered.

This will be solved when using LoRaWAN Join Servers, which is something I would recommend the manufacturer doing.

mazareth, 2017-11-15:
Thanks johan,
I will contact the manufacturer.

Another solution could be do to a personal lora server?

johanstokking, 2017-11-16:
Not really, because join requests may conflict then.

Manufacturers should not ship devices that can't change security settings and that are already registered with their AppEUI, DevEUI and AppKey combination.