TheThingsNetwork/docs #177: How saving device message

mazareth, 2017-11-20:
Hi,

can we have an historic of device message?

When i leave the thethingnetwork interface. All messages are lose.
But i want to test my gateway during a week.

Have you an integrated solution?

johanstokking, 2017-11-20:
@mazareth yes. We have the Storage Integration, see https://www.thethingsnetwork.org/docs/applications/storage/

closed 2017-11-20 by johanstokking

mazareth, 2017-11-20:
Thanks.

Thats what i searching for.

mazareth, 2017-11-21:
Hi,
I've configure the data storage yesterday but this morning i've only the 3 last data saved on my GUI.

[image]

When i take the data it's ok
[image]
[image]

But how can i see them in thethingnetwork GUI?

johanstokking, 2017-11-22:
@mazareth we don't go back that far in the Console; it's set to 30 minutes. If you need more history, call the API directly.

mazareth, 2017-11-23:
What do you mean by calling the API?

I've to install curl?
I don't know how to get the data and have a friendly interface.

johanstokking, 2017-11-23:
Friendly interfaces are provided by many IoT platforms. See the Integrations panel in the Application Overview. You can use Cayenne, AllThingsTalk, EVRYTHNG, etc, all providing visualization and long time data storage. Also, it's easy to build something yourself with Node-RED. We focus specifically on the networking layer.

mazareth, 2017-11-24:
Thanks again Johan.
I test Cayenne and EVRYTHING. 
Cayenne works and give me what i want. It's easy to use.

I don't understand how EVRYTHING work.