TheThingsNetwork/docs #190: wrong Name of Temperature Sensor in "node-reset-led.jpg"

8nich, 2018-01-17:
There is a typo in the name of the temperature sensor. It states it is the MCP9204 (in the picture:  [https://github.com/TheThingsNetwork/docs/blob/master/_content/devices/node/node-reset-led.jpg](url)) but instead it's the: MCP9804 (MCP9204 does not exist)

johanstokking, 2018-01-17:
Hi @8nich, thanks for your contribution. The link you posted gives a 404. Can you file a pull request with the fix?

8nich, 2018-01-17:
@johanstokking sorry about that. Here the picture from the ttn page, which is live. https://www.thethingsnetwork.org/docs/devices/node/node-reset-led.jpg 
Maybe it's easier, if you or the one which has drawn the picture, modifies it. It would also have impact on the text shown at the distributors page: https://uk.rs-online.com/web/p/sensor-development-kits/1359784/

johanstokking, 2018-01-18:
Ah now I see, thanks.

closed 2021-03-22 by nejraselimovic