TheThingsNetwork/docs #240: Add guide to contribute to the documentation

MaartenArits, 2018-04-10:
I would like to contribute to the documentation but is don't know how to do this. A reference to the tools used and their getting started guides would be very helpful. I suggest doing this in the Readme.md file or at least point in the right direction

johanstokking, 2018-04-11:
Great!

What misses exactly from the README? It guides you running everything locally.

The rest is basically the GitHub workflow; fork the repository, make changes (in a new branch), and submit a pull request here.

MaartenArits, 2018-04-11:
I'm not yet to familiar with the work flow but would love to contribute. I guess is have to start by learning Jekyll? Is there a standard way to order documents, upload images, add JavaScript and so on or does it all become clear once you get started?

johanstokking, 2018-04-11:
Yes, it's essentially a matter of working under `_content` and following in the instructions in the README.

closed 2018-04-11 by johanstokking