TheThingsNetwork/docs #284: docs: the things node quick start wrong intervals?

tdsmtdsm, 2018-12-15:
The [quick start doc](https://github.com/TheThingsNetwork/docs/blob/master/_content/devices/node/quick-start.md) for TheThingsNode [mentions](https://www.thethingsnetwork.org/docs/devices/node/quick-start.html#interval) 8 seconds and 8000ms as sleep interval but the Arduino library example code contains 60000 as sleep interval.

johanstokking, 2018-12-16:
Thanks for reporting.

>but the Arduino library example code contains 60000 as sleep interval.

Can you add a reference?

tdsmtdsm, 2018-12-16:
The Basic example sketch has 60000 as interval argument: [https://github.com/TheThingsNetwork/arduino-node-lib/blob/master/examples/Basic/Basic.ino#L56](https://github.com/TheThingsNetwork/arduino-node-lib/blob/master/examples/Basic/Basic.ino#L56)

Elsewhere, the quick start doc talks about 20 seconds as interval: [https://github.com/TheThingsNetwork/docs/blob/master/_content/devices/node/quick-start.md#send-data](https://github.com/TheThingsNetwork/docs/blob/master/_content/devices/node/quick-start.md#send-data)

Might be confusing for quick starters.

closed 2021-03-22 by nejraselimovic