TheThingsNetwork/docs #330: Bundler could not find compatible versions for gem "ruby ":

MarkNambatac, 2019-10-20:
When I run the command `bundle install`, this is what i get: 

```
Fetching gem metadata from https://rubygems.org/..........
Fetching gem metadata from https://rubygems.org/.
Resolving dependencies...
Bundler could not find compatible versions for gem "ruby ":
  In Gemfile:
    ruby

    jemoji was resolved to 0.9.0, which depends on
      html-pipeline (~> 2.2) was resolved to 2.7.1, which depends on
        nokogiri (>= 1.4) was resolved to 1.8.2, which depends on
          ruby  (< 2.6) x64-mingw32

Could not find gem 'ruby  (< 2.6)', which is required by gem 'nokogiri (>=
1.4)', in any of the relevant sources:
  the local ruby installation
```

I found a solution by running `bundle update`, but this will create an error when I push it in github: 
[image]

I am using **ruby 2.6.4p104** (2019-08-28 revision 67798)

johanstokking, 2019-10-22:
@MathieuMonneret can you pick this up as part of dependency updates?

MathieuMonneret, 2019-10-31:
We updated travis CI configuration to work with latest Bundler version, 
could you try pushing it to git once more and tell us if you experiment the same issue.

johanstokking, 2019-10-31:
Please reopen if this issue persists.

closed 2019-10-31 by johanstokking