TheThingsNetwork/docs #379: AWS IoT Quickstart Integrations fails

sprobst, 2020-05-07:
Creation of a new Beanstalk stack fails with:
`No Solution Stack named '64bit Amazon Linux 2018.03 v2.15.1 running Go 1.14.1' found. (Service: AWSElasticBeanstalk; Status Code: 400; Error Code: InvalidParameterValue; Request ID: <request id omitted>)`
I am using the template link from [documentation](https://www.thethingsnetwork.org/docs/applications/aws/quick-start.html). I tried to create the stack in Frankfurt (eu-central-1) and Ireland (eu-west-1).

Furthermore, a sync interval of five minutes cannot be selected.

johanstokking, 2020-05-07:
Thanks for reporting. The build is running for the new solution stack. Expect it to be ready in ~10 minutes.

closed 2020-05-07 by johanstokking

sprobst, 2020-05-07:
It is working. Thank you for your quick response!