TheThingsNetwork/docs #400: Edit on GitHub links - how should I fix?

shawaj, 2020-10-01:
So I have noticed on quite a few pages on the Docs, the "Edit api.md" (or other file) link at top right linking to github references the old location of the docs due to a hard-coded "source" declaration. For example here:

https://github.com/TheThingsNetwork/docs/blob/master/_content/applications/evrythng/index.md

As you can see the source tag wrongly references this old location:

https://github.com/TheThingsIndustries/integration-messaging/blob/master/evrythng/README.md

I was going to fix this to make it all work nicely, but I wanted to check which of the following options would be preferred:

1. Change the code at https://github.com/TheThingsNetwork/docs/blob/master/_layouts/page.html#L46 in order to ignore any "source" declaration and simply use the github page path (i.e. remove `{% if page.source %}{{ page.source }}`)

2. Go through any pages with an incorrect source declaration and fix them individually / remove the override

I am thinking 2 is the best option as it is cleaner and still allows for manual overrides of the source link. However I just wasn't sure if there was actually any scenarios you envisage where you would have a source URL which is different to the page path?

Could you let me know which would be preferred and I will get it all updated?

shawaj, 2020-10-12:
@johanstokking @wienke any thoughts on this?

I don't mind fixing it but just want to make sure I am doing it how you would like it.

johanstokking, 2020-10-13:
@shawaj thanks for your willingness to contribute on this.

The thing is that the source file is in fact in a private repository. But indeed it doesn't make a lot of sense to have an edit button for those files. I'm not sure if we can differentiate easily.

Now, the thing with this documentation is that we're moving away from this and we don't maintain the technicalities behind these docs. So I wouldn't bother with this anymore, unless you have an easy fix!

nejraselimovic, 2021-03-22:
Closing this as it is not relevant anymore.

closed 2021-03-22 by nejraselimovic