TheThingsNetwork/docs #406: Mark docs as V2 only and link to V3 documentation

benolayinka, 2020-11-02:
Let's mark this documentation as V2 specific and link to the V3 docs for The Things Stack.

Here:
[image]

And here:
[image]

benolayinka, 2020-11-03:
@GreenPenguino have you worked on the TTN docs page? Could you help modify the layout to add this note?

GreenPenguino, 2020-11-03:
I have not worked on TTN docs, but willing to try

benolayinka, 2020-11-03:
Will be closed by https://github.com/TheThingsNetwork/docs/pull/407

PR #407 (2020-11-11): Add v3 docs note
Closes #406. Draft to add a v2/v3 note to the TTN docs. @johanstokking can you give some feedback about the text? @GreenPenguino can you double check the HTML?

Looks like:

[image]

And:

[image]
closed 2020-11-11 by benolayinka