TheThingsNetwork/docs #434: Landing page icons redesign

nejraselimovic, 2021-03-16:
We need some redesigned icons for the TTN docs landing page. This is what we currently have:

[image]
[image]

I guess we can reuse some icons used for TTS docs page, i.e.:

[image]

@benolayinka @laurensslats any ideas?

pierrephz, 2021-03-24:
@nejraselimovic find here a schema you can use for the Learn page:

[image]

Here is a .zip where you will find icons to replace the outdated ones: [learn_page_icons.zip](https://github.com/TheThingsNetwork/docs/files/6198618/learn_page_icons.zip)

Later this week all the icons we have so far will be available in the design repo

pierrephz, 2021-03-24:
Quick UI fix:

[image]

If you can just put the text on top of the shema (and not beside) it would be cool.
I see this page needs a complete UI rework but I don't know when I will be able to dot it, so until then it could be nice to do this

nejraselimovic, 2021-03-24:
> Quick UI fix:
> 
> [image]
> 
> If you can just put the text on top of the shema (and not beside) it would be cool.
> I see this page needs a complete UI rework but I don't know when I will be able to dot it, so until then it could be nice to do this

Sure, that should be easy to fix. Great work with the icons!

benolayinka, 2021-08-05:
Closed by #467

closed 2021-08-05 by benolayinka