TheThingsNetwork/docs #445: Update documentation to match the branding

nejraselimovic, 2021-04-01:
#### Summary
We decided to use:

- For V2 -> The Things Network V2 for community deployment, The Things Industries V2 for private deployments
- For V3 -> The Things Stack <deployment name> and only mention that it is the version 3 (V3) somewhere in the beginning

so the whole documentation needs to reflect this decision.

#### Why do we need this ?
To avoid confusion with branding

#### Can you do this yourself and submit a Pull Request?
Yes, but can you @benolayinka and @laurensslats please confirm this is correct before I start making changes

jpmeijers, 2021-04-01:
As an end user I would want to see the branding on the console match whatever is used in the documentation:
[image]

jpmeijers, 2021-04-01:
I've seen @laurensslats is starting to use the name "The Things Stack Community" for the version I as a community member use. Shouldn't that be "The Things Stack Community Edition", because "The Things Stack Community" sounds like you are referring to the people - the people that makes up the community. Where as "Community Edition" is clear that it is the instance that is meant for the people in the community.

nejraselimovic, 2021-04-07:
Closed with #443.

closed 2021-04-07 by nejraselimovic