TheThingsNetwork/docs #456: Change the UI of the "What's new" section

pierrephz, 2021-06-07:
#### Summary

Change the UI of the `What's new` section

#### Why do we need this ?

Now, the UI tells the user that he can scroll horizontally. But this is not the case, so we need to fix this

#### What is already there? What do you see now?

[image]

#### What is missing? What do you want to see?

[image]

Remove this gradient on the right side, add the number of changes that have been made

#### How do you propose to document this?

@benolayinka can you take this up? or let me know how you want to proceed
The mockup is ready, I can give further details to the developer

#### Can you do this yourself and submit a Pull Request?

Need @benolayinka

benolayinka, 2021-06-08:
dunno if this is worth it but @kschiffer let me know what you think. if you like I can also implement it, you'll just be faster and produce a better result 🤷🏼‍♀️

kschiffer, 2021-06-08:
I think @pierrephz is giving this a go himself actually. :muscle:

closed 2021-07-07 by pierrephz