TheThingsNetwork/docs #473: Please compose a glossary page with a comprehensive list of names used to name to the newer version of The Things Network

HouzuoGuo, 2021-09-19:
#### Summary

There are a vast number of terms currently in-use to name the *newer* version of The Things Network:

- simply "The Things Network"
- The Things Network Stack
- The Things Stack
- The Things Stack v3
- The Things Stack Community Edition
- The Things Network v3

All of these terms have been in-use, largely interchangeably, across The Things Stack documentation, forum posts from the developers of The Things Network, and reputable forum users. 

Therefore, please create a new glossary page that clearly states the meaning of these interchangeable terms. This will give new users a far clearer picture of what the new network really is.

#### Can you do this yourself and submit a Pull Request?

Not yet.

jpmeijers, 2021-09-20:
I agree with you. Even though there are official names we should use, those are long and hard to type. Therefore many of the old user like me end up calling the new version something like TTNv3.

This issue might be of interest too: https://github.com/TheThingsNetwork/docs/issues/445

descartes, 2022-08-26:
Been there on the stack repro, here are the answers given:

TTS CE = TTN
TTS OS = you hosting your own stack for free

I'll have to dig out what the TTS paid for version is called.

HouzuoGuo, 2024-01-27:
(Copying my remark over from #547)

This is not a general comment, but a legit technical challenge highlighting the shortcomings of documentation. Nowhere in the documentation is there a list of glossaries of many historical and current names of the newer version of the things network, the prior issue remains unresolved till this day.

Please get in touch with your branding and marketing colleagues to determine an appropriate course forward. Otherwise, documentation readers - developers and users alike, currently have minimal clue whether any technical documentation page applies to the newer version of the things network.

nejraselimovic, 2024-02-04:
Hi @HouzuoGuo, thanks for reporting, I will pick this up and try to remove any existing confusion about our products' names.