TheThingsNetwork/docs #498: Document additional LoRa hardware

benolayinka, 2022-01-05:
#### Summary

Once #497 is merged, lets add more common devices from https://www.semtech.com/products/wireless-rf/lora-core and the LR1110 in the same format.

#### Why do we need this ?

Provide additional information on TTN learn

nejraselimovic, 2022-01-06:
do we have any info on which devices exactly are the common ones?

jpmeijers, 2022-01-06:
* SX1276, SX1272 - supported by LMiC - https://github.com/mcci-catena/arduino-lmic

* SX1261, SX1262 and SX1268 - supported by https://github.com/beegee-tokyo/SX126x-Arduino

* And then the LR1110.

These are the only ones I've really seen being used. Even the SoCs and modules that can do LoRa are all based on one of the above ones.

nejraselimovic, 2022-01-06:
> * SX1276, SX1272 - supported by LMiC - https://github.com/mcci-catena/arduino-lmic
>
>     * SX1261, SX1262 and SX1268 - supported by https://github.com/beegee-tokyo/SX126x-Arduino
>
>     * And then the [...]

Great, thanks a lot for the super useful info!

PR #500 (2022-02-16): Document common LoRa devices
#### Summary
Closes #498 

#### Notes for Reviewers
I checked datasheets for each one of these chips and found out that 90% of the info about chips from the same series is the same, so I aggregated the info to describe the whole series instead of chip by chip and just highlighted differences where they exist

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links.
- [x] Style Guidelines: Documentation obeys style g [... cut]
closed 2022-02-16 by nejraselimovic