TheThingsNetwork/docs #501: Strikethrough support in Markdown

loleg, 2022-01-10:
It would be quite useful to be able to use the full range of Markdown in the community [events](https://www.thethingsnetwork.org/community/Bern/post/new-years-meetup-12012022) section of the website. Currently the editor shows the strikethrough (`~~some text~~`) formatting in the preview, but does not render it accordingly in the published page.

benolayinka, 2022-01-11:
@michalborkowski96

michalborkowski96, 2022-01-28:
@loleg 

I have submitted a PR for this issue. Once it gets accepted, merged and deployed, an ~~S~~ button for strike-through should appear in the editor. Once you see it, please edit and re-save your section, and the strike-through should work.

Realistically, this should be deployed somewhere next week, although that's not a promise.

closed 2022-02-04 by michalborkowski96

michalborkowski96, 2022-02-04:
The change has been deployed

loleg, 2022-02-07:
Tested and working as promised. Many thanks! Now what else can I strike out...