TheThingsNetwork/docs #504: Distinguish better between device registration and activation

furtiman, 2022-02-04:
#### Summary

The sequence of pages that describe process of registering, and then activating the device use confusing wording.
1. https://www.thethingsindustries.com/docs/getting-started/cloud-hosted/tti-join-server/register-devices/
2. https://www.thethingsindustries.com/docs/getting-started/cloud-hosted/tti-join-server/activate-devices-cloud-hosted/

#### Why do we need this ?

...

#### What is already there? What do you see now?

The sequence can be accessed from [Make a Device Claimable](https://www.thethingsindustries.com/docs/devices/device-claiming/make-device-claimable/) page, that belongs to the docs intended for device makers.

The word “register” is used in both of them, however one page is “[register](https://www.thethingsindustries.com/docs/getting-started/cloud-hosted/tti-join-server/register-devices/)” (where `ttn-lw-cli end-devices create test-app test-device` is used, and the page ends with “You can now activate the devices on The Things Industries Cloud Hosted”)

that leads to another that is about [activation](https://www.thethingsindustries.com/docs/getting-started/cloud-hosted/tti-join-server/activate-devices-cloud-hosted/) (where `ttn-lw-cli end-devices set test-app test-device` is used, and the page also ends with “You can now activate the devices on The Things Industries Cloud”)

#### What is missing? What do you want to see?

If no other actions are required in this sequence, then the latter page should sound more like “activate with … -> Now it’s activated”

It should also link back to the [Make a Device Claimable](https://www.thethingsindustries.com/docs/devices/device-claiming/make-device-claimable/) page, which tells about further steps in the process

#### How do you propose to document this?

Described above, plus the wording "register" and "activate" should be thoroughly checked in the content of both page to appear less confusing

#### Can you do this yourself and submit a Pull Request?

-

nejraselimovic, 2022-02-11:
looping in @johanstokking

johanstokking, 2022-02-22:
We're going to change this entirely with The Things Join Server. All this stuff will be gone. I'm closing the issue because we track this work elsewhere, but also because this issue is not in the right repo.

closed 2022-02-22 by johanstokking