TheThingsNetwork/docs #507: Update Maximum Duty Cycle information as per the EU regulations

ymgupta, 2022-02-23:
#### Summary
The duty cycles regulations in Europe are defined in **Table B.1** of the [ETSI EN 300 220-2 V3.2.1](https://www.etsi.org/deliver/etsi_en/300200_300299/30022002/03.02.01_60/en_30022002v030201p.pdf) standard.
[image]

#### Why do we need this ?
To correct the duty cycle regulations of the `g band` in Europe as per the regulations.
#### What is already there? What do you see now?
The documentation [here](https://www.thethingsnetwork.org/docs/lorawan/duty-cycle/#maximum-duty-cycle) states that the duty cycle for the `g band` is `1%`, but according to [regulations](https://www.etsi.org/deliver/etsi_en/300200_300299/30022002/03.02.01_60/en_30022002v030201p.pdf), it is about 0.1%.

> g (863.0 – 868.0 MHz): 1%

#### What is missing? What do you want to see?
Update the Duty Cycle of `g band` in the documentation according to the [EU regulations](https://www.etsi.org/deliver/etsi_en/300200_300299/30022002/03.02.01_60/en_30022002v030201p.pdf).
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No.

nejraselimovic, 2023-02-26:
@pradeeka7 does your [PR](https://github.com/TheThingsNetwork/docs/pull/530) address this too?

pradeeka7, 2023-02-27:
I could not find the 'g-g4' naming scheme in this [document](https://www.etsi.org/deliver/etsi_en/300200_300299/30022002/03.02.01_60/en_30022002v030201p.pdf), so we agreed to use the naming scheme (K-Q) specified in Table B.1." We now have to list 6 sub-bands instead of 5, as stated on the [Duty Cycle](https://www.thethingsnetwork.org/docs/lorawan/duty-cycle/) page. I verified it with @adriansmares. 

@ymgupta If we are wrong please correct us.

[image]

PS - I still can't understand where the naming scheme 'g-g4' comes from, and all sources on the internet refer to it in the context of the [ETSI EN 300 220-2 V3.2.1 (2018-06) document](https://www.etsi.org/deliver/etsi_en/300200_300299/30022002/03.02.01_60/en_30022002v030201p.pdf).
After further searching, I found a [document ](https://www.etsi.org/deliver/etsi_tr/103000_103099/103055/01.01.01_60/tr_103055v010101p.pdf)that mentioned frequency bands 'g-g4,' but it does not seem to match with LoRaWAN.

Page 10:
[image]

pradeeka7, 2023-04-30:
@benolayinka, @nejraselimovic, @adriansmares please help me clarify this. I am still confused about these European sub-bands.

adriansmares, 2023-05-01:
The G/G1-G4 naming scheme is a legacy naming scheme which I don't believe is in use any more. It is very likely that the old docs used it out of inertia (someone wrote them in 2014-2016, and then we've been carrying them ever since). The document linked about this scheme has been published in ~2011.

The `ETSI EN 300 220-2` is published in ~2018 and is leading in this case. We use the K-Q band naming scheme, as well as restrictions, in The Things Stack today.

pradeeka7, 2023-05-01:
@adriansmares - Thanks for the clarification. I used this K-Q naming scheme in this article (https://www.thethingsnetwork.org/docs/lorawan/regional-limitations-of-rf-use/). I remember we discussed the same for that.

I will update the page https://www.thethingsnetwork.org/docs/lorawan/frequency-plans/ and will submit a PR.

pradeeka7, 2023-05-02:
@NicolasMrad can you assign this to me?

pradeeka7, 2023-05-25:
@NicolasMrad - Now you can close this :)

closed 2023-05-25 by NicolasMrad