TheThingsNetwork/docs #540: Large volume of text in Regional Parameters page

pradeeka7, 2023-08-10:
#### Summary

The regional Parameters page has a large volume of text and needs to be reduced.
https://www.thethingsnetwork.org/docs/lorawan/regional-parameters/

#### Why do we need this ?

To find key parameters for each region easily.

#### What is already there? What do you see now?

The regional parameters that are required for The Things Certification.

#### What is missing? What do you want to see?

...

#### How do you propose to document this?

Remove the text that is not required for the exam.

#### Can you do this yourself and submit a Pull Request?

Yes, @nejraselimovic can you assign this to me?

jpmeijers, 2023-08-10:
> Remove the text that is not required for the exam.

I agree the page is long and has a lot of text, but just because the info is not in the exam doesn't make worthless. Rather just split the page into multiple pages, and keep the details. The community, including me and ex-TTI employees spent a lot of time gathering all this info. Throwing all that away just because it's not in the exam (which people like me helped set), is not a good idea.

pradeeka7, 2023-08-12:
This issue is originally mentioned in:
https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/1143

> I agree the page is long and has a lot of text, but just because the info is not in the exam doesn't make worthless.

Me too. Let's split this into sub pages.

> The community, including me and ex-TTI employees spent a lot of time gathering all this info. Throwing all that away just because it's not in the exam (which people like me helped set), is not a good idea.

The content of this page, starting from the heading '**Common Frequency Plans**' was originally written by me about 2 years ago. I am fully aware of the content of The Things Certification. I am one of the main contributors, along with Laurens and Remko, for The Things Certification. 🙂

nejraselimovic, 2023-08-12:
> I agree the page is long and has a lot of text, but just because the info is not in the exam doesn't make worthless. Rather just split the page into multiple pages, and keep the details. The [...]

I actually agree with this. Why remove useful content? It can be properly reorganized, and parts that are important for the certification can be higlighted as such.

closed 2024-01-15 by pradeeka7