TheThingsNetwork/docs #547: Please stop renaming the community network, which is already known by 7 different names at this point.

HouzuoGuo, 2024-01-26:
See also prior issue: https://github.com/TheThingsNetwork/docs/issues/473

#### Summary
The community-run free edition of your network is already known by 6(!) largely interchangeable names, this is incredibly confusing and incredibly harmful for developers and users.

Last I logged into the network console, it's been renamed again for the 7th time to "The Things Stack Sandbox".

Please stop this madness.

KrishnaIyer, 2024-01-26:
Please use the forum for general comments.

closed 2024-01-26 by KrishnaIyer

HouzuoGuo, 2024-01-27:
This is not a general comment, but a legit technical challenge highlighting the shortcomings of documentation. Nowhere in the documentation is there a list of glossaries of many historical and current names of the *newer* version of the things network, the prior issue remains unresolved till this day.

Please get in touch with your branding and marketing colleagues to determine an appropriate course forward. Otherwise, documentation readers - developers and users alike, currently have minimal clue whether any technical documentation page applies to the *newer* version of the things network.