TheThingsNetwork/docs #554: Docs update: Frequency Plans by Country

ticlatam, 2024-07-03:
#### Summary

Update or Delete the country list that is incomplete and outdated and keep only the link to the LoRa Alliance documentation that has an updated version. If deleted, a drawback is that the "Regulatory document" column will be lost because is not detailed in the LoRa Alliance Regional Parameters document. Also Lora Alliance document in some countries has two available options.

#### Why do we need this ?

Wrong data may cause several issues.

#### What is already there? What do you see now?

[image]
For example, several south american countries use AU915 instead of US915: Argentina, Brasil, Chile, Colombia, República Dominicana, Ecuador, El Salvador, Guatemala, Honduras, Nicaragua, Panamá, Paraguay, Perú y Uruguay.

#### What is missing? What do you want to see?

#### How do you propose to document this?

Revamp the list considering the [LoRa Alliance](https://resources.lora-alliance.org/technical-specifications/rp002-1-0-4-regional-parameters) documentation. Even [Helium](https://docs.helium.com/iot/lorawan-region-plans/) has an updated one...

#### Can you do this yourself and submit a Pull Request?

Recently I updated Ecuador's data but I noticed other countries need to be updated as well. I can try to update it but how to determine which list is "the one" to use? What do you think @nejraselimovic ?

jpmeijers, 2024-07-05:
Being a little subjective here, as I was the original author for this list when it was still on the TTN Wiki (circa April 2017). Helium at some point copied our list verbatim. That indicates they agree that having a clear list like this is beneficial.

I'd like to see this list being updated, mainly using the LoRaWAN Regional Parameters document as source, but adding references where we can.

The original idea behind this was to publish a partial list of countries that are well-known, along with their documents as reference, but letting the community add where they can.

ticlatam, 2024-07-05:
Ok, so I can add the other countries that the LoRaWAN Regional Paramenters has that TTN is missing, and update some countries as well. What do you think about writing a sentence about how to suggest editions because the last paragraph only mention that  discussions about the frequency plans in different countries takes place on the Forum, but nothing about this Github.

nejraselimovic, 2024-07-06:
hi @ticlatam, thanks for reporting this. This definitely needs to be updated. 

@pradeeka7 can you check it out? I know you've been working on Regional Parameters doc recently. If not, I can pick it up.

This is also a nice suggestion.
> What do you think about writing a sentence about how to suggest editions because the last paragraph only mention that discussions about the frequency plans in different countries takes place on the Forum, but nothing about this Github.